electrified/asus-wmi-sensors #86: Support for MAXIMUS VII HERO

nknotzer, 2024-03-03:
Hi,

MAXIMUS VII HERO is supported by HWINFO. IMHO it should be possible to add support for this MB. 

Output of dmidecode:

Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: MAXIMUS VII HERO
	Version: Rev 1.xx
	Serial Number: 141033558201078
	Asset Tag: To be filled by O.E.M.
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: To be filled by O.E.M.
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

It would be great, if you could add support for this MB. 

Thank you!

0lvin, 2024-03-03:
Hi, 

This motherboard does not have a known signatures of wmi endpoints in available bios [dumps](https://github.com/asus-wmi-boards-sensors/asus-board-dsdt/blob/master/README.md).

In theory it could be supported by nct6775 kernel modules as has NCT6791D chip.

Could you check what chip you have on board and what messages provided in dmesg when you load nct67** kernel module?

If it has something about chip detected but has resource conflict, you could also try to add your board to asus_wmi_boards list in [module](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/hwmon/nct6775-platform.c?h=v6.7.8).

nknotzer, 2024-04-15:
hi, 

the MB has the NCT6791D chip (see kernel dump). It has the known issue in respect to the ACPI resource conflict (https://askubuntu.com/questions/875323/unable-to-get-lm-sensors-to-report-my-fan-speeds-for-asus-z97-ws)

Apr 15 17:24:18 pc-nknotzer kernel: nct6775: Enabling hardware monitor logical >
Apr 15 17:24:18 pc-nknotzer kernel: nct6775: Found NCT6791D or compatible chip >
Apr 15 17:24:18 pc-nknotzer kernel: ACPI Warning: SystemIO range 0x000000000000>
Apr 15 17:24:18 pc-nknotzer kernel: ACPI: OSL: Resource conflict; ACPI support >