electrified/asus-wmi-sensors #87: Board not supported, not capable of modprobe

Aman1994, 2024-04-11:
I was trying to compile the driver but while doing so i am getting 
```
modprobe: ERROR: could not insert 'asus_wmi_sensors': No such device
```

This is the motherboard:

```
Manufacturer: ASUSTeK COMPUTER INC.
Product Name: PRIME X670-P WIFI
```

It would be great, if you could add support for this MB.

Thank you!

KeithMyers, 2024-04-11:
Your board is not supported.
You need to try the alternate driver asus-ec-sensors which is provided by the latest kernels.
Or you need to ask for assistance on its Github repo.
[https://github.com/zeule/asus-ec-sensors](https://github.com/zeule/asus-ec-sensors)

0lvin, 2024-04-11:
Nct6775 has support of your board, look to https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hwmon/nct6775-platform.c#n1319.

Aman1994, 2024-04-12:
> Your board is not supported. You need to try the alternate driver asus-ec-sensors which is provided by the latest kernels. Or you need to ask for assistance on its Github repo. https://github.com/zeule/asus-ec-sensors

Hi @KeithMyers  , i tried using https://github.com/zeule/asus-ec-sensors - but that too doesn't work

```
:~/asus-ec-sensors:master:v0.1.0-54-ge763a87
# make dkms
Creating symlink /var/lib/dkms/asus-ec-sensors/v0.1.0-54-ge763a87/source -> /usr/src/asus-ec-sensors-v0.1.0-54-ge763a87
Sign command: /lib/modules/4.18.0-513.18.1.el8_9.x86_64/build/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
make -j32 KERNELRELEASE=4.18.0-513.18.1.el8_9.x86_64 KVER=4.18.0-513.18.1.el8_9.x86_64 DKMS_BUILD=1...
Signing module /var/lib/dkms/asus-ec-sensors/v0.1.0-54-ge763a87/build/asus-ec-sensors.ko
Cleaning build area...

asus-ec-sensors.ko.xz:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.18.0-513.18.1.el8_9.x86_64/extra/
Adding any weak-modules
depmod...
modprobe: ERROR: could not insert 'asus_ec_sensors': Required key not available
make: *** [Makefile:59: dkms] Error 1
```

Aman1994, 2024-04-12:
> Nct6775 has support of your board, look to https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hwmon/nct6775-platform.c#n1319.
> 
> If it's not true could you please attach dmesg messages when you load nct6775 to https://bugzilla.kernel.org/show_bug.cgi?id=204807.

Hello @0lvin ,

I can see the module is present 

```
modinfo nct6775
filename:       /lib/modules/4.18.0-513.18.1.el8_9.x86_64/kernel/drivers/hwmon/nct6775.ko.xz
license:        GPL
description:    Driver for NCT6775F and compatible chips
author:         Guenter Roeck <linux@roeck-us.net>
rhelversion:    8.9
srcversion:     614D3F059EC58795E8F119A
depends:        hwmon-vid
intree:         Y
name:           nct6775
vermagic:       4.18.0-513.18.1.el8_9.x86_64 SMP mod_unload modversions 
sig_id:         PKCS#7
signer:         Red Hat Enterprise Linux kernel signing key
sig_key:        3A:33:E5:95:F0:BE:59:3D:1E:91:14:71:D3:D1:4D:47:C7:EC:4D:78
sig_hashalgo:   sha256
signature:      0F:25:C8:01:F7:0A:A5:29:8E:57:16:8E:12:20:C4:62:69:DE:92:33:
		63:25:B7:9E:9D:CA:14:3E:C2:37:8B:C1:76:7F:62:2F:FC:93:37:14:
		87:F9:3A:20:D7:59:88:AD:13:D7:A0:3E:5B:F3:C7:99:00:B2:AE:2D:
		20:02:DC:A5:C5:F5:33:B8:59:C2:23:B8:85:40:0D:D5:B0:78:53:DD:
		42:99:D2:BB:40:C8:03:8F:CD:EA:4A:1E:0C:0A:BE:EC:19:04:57:90:
		1A:1C:FA:AD:B0:65:83:11:6E:8A:E0:8C:6D:DB:0F:FC:48:80:08:7C:
		69:A0:BF:F0:23:2D:7A:EE:C5:CB:A5:50:27:6A:EC:52:87:8C:23:A7:
		26:CB:9D:07:F6:6D:99:03:7F:9F:C0:DD:0B:0C:4D:B6:5E:7F:52:38:
		B0:9D:B4:70:7A:AD:AC:AB:E1:AA:FD:BA:BE:91:85:80:24:4E:7F:02:
		1A:66:31:9E:7D:FD:DC:13:F6:F2:8F:D2:84:AC:13:FC:2E:DF:D8:3E:
		22:34:20:A7:D9:3E:D8:7C:73:3A:E4:3F:3B:0B:BD:90:6F:65:40:2C:
		B1:3C:89:E2:DE:83:D1:D7:D9:58:69:03:1D:86:16:2C:15:24:34:B0:
		E5:37:C2:7A:9E:DB:8C:BF:C8:1D:C3:8A:33:53:C0:4B:A2:67:70:A2:
		D5:EC:E5:2B:45:A0:CC:EB:56:4D:40:DF:34:F3:E8:71:0C:92:CC:59:
		83:89:C0:F9:BA:10:C4:1F:A8:E8:51:D7:73:0C:50:E9:E6:95:96:F0:
		8B:CF:4F:73:19:09:BE:A2:33:70:51:D4:E7:A6:22:C7:41:30:9C:5A:
		6E:48:20:16:32:66:44:41:BA:48:9B:9C:4D:7E:04:5D:08:55:04:5A:
		5E:80:29:12:40:30:5D:59:B3:59:A4:F1:1D:24:C5:08:AF:F1:8D:E9:
		5E:B9:33:08:9A:6F:65:B0:E9:51:90:1F:66:A9:00:DE:3C:20:DB:EB:
		C7:BD:79:CE
parm:           force_id:Override the detected device ID (ushort)
parm:           fan_debounce:Enable debouncing for fan RPM signal (ushort)

```
But, when i try to load that module, it doesn't get loaded 

```
modprobe nct6775
modprobe: ERROR: could not insert 'nct6775': No such device

```

0lvin, 2024-04-12:
> > Nct6775 has support of your board, look to https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/hwmon/nct6775-platform.c#n1319.
> > If it's not true could you please attach dmesg messages when you load nct6775 to https://bugzilla.kernel.org/show_bug.cgi?id=204807.
> 
> Hello @0lvin ,
> 
> I can see the module is present
> 
> ```
> modinfo nct6775
> filename:       /lib/modules/4.18.0-513.18.1.el8_9.x86_64/kernel/drivers/hwmon/nct6775.ko.xz
> license:        GPL
> description:    Driver for NCT6775F and compatible chips
> author:         Guenter Roeck <linux@roeck-us.net>
> rhelversion:    8.9
> srcversion:     614D3F059EC58795E8F119A
> depends:        hwmon-vid
> intree:         Y
> name:           nct6775
> vermagic:       4.18.0-513.18.1.el8_9.x86_64 SMP mod_unload modversions 
> sig_id:         PKCS#7
> signer:         Red Hat Enterprise Linux kernel signing key
> sig_key:        3A:33:E5:95:F0:BE:59:3D:1E:91:14:71:D3:D1:4D:47:C7:EC:4D:78
> sig_hashalgo:   sha256
> signature:      0F:25:C8:01:F7:0A:A5:29:8E:57:16:8E:12:20:C4:62:69:DE:92:33:
> 		63:25:B7:9E:9D:CA:14:3E:C2:37:8B:C1:76:7F:62:2F:FC:93:37:14:
> 		87:F9:3A:20:D7:59:88:AD:13:D7:A0:3E:5B:F3:C7:99:00:B2:AE:2D:
> 		20:02:DC:A5:C5:F5:33:B8:59:C2:23:B8:85:40:0D:D5:B0:78:53:DD:
> 		42:99:D2:BB:40:C8:03:8F:CD:EA:4A:1E:0C:0A:BE:EC:19:04:57:90:
> 		1A:1C:FA:AD:B0:65:83:11:6E:8A:E0:8C:6D:DB:0F:FC:48:80:08:7C:
> 		69:A0:BF:F0:23:2D:7A:EE:C5:CB:A5:50:27:6A:EC:52:87:8C:23:A7:
> 		26:CB:9D:07:F6:6D:99:03:7F:9F:C0:DD:0B:0C:4D:B6:5E:7F:52:38:
> 		B0:9D:B4:70:7A:AD:AC:AB:E1:AA:FD:BA:BE:91:85:80:24:4E:7F:02:
> 		1A:66:31:9E:7D:FD:DC:13:F6:F2:8F:D2:84:AC:13:FC:2E:DF:D8:3E:
> 		22:34:20:A7:D9:3E:D8:7C:73:3A:E4:3F:3B:0B:BD:90:6F:65:40:2C:
> 		B1:3C:89:E2:DE:83:D1:D7:D9:58:69:03:1D:86:16:2C:15:24:34:B0:
> 		E5:37:C2:7A:9E:DB:8C:BF:C8:1D:C3:8A:33:53:C0:4B:A2:67:70:A2:
> 		D5:EC:E5:2B:45:A0:CC:EB:56:4D:40:DF:34:F3:E8:71:0C:92:CC:59:
> 		83:89:C0:F9:BA:10:C4:1F:A8:E8:51:D7:73:0C:50:E9:E6:95:96:F0:
> 		8B:CF:4F:73:19:09:BE:A2:33:70:51:D4:E7:A6:22:C7:41:30:9C:5A:
> 		6E:48:20:16:32:66:44:41:BA:48:9B:9C:4D:7E:04:5D:08:55:04:5A:
> 		5E:80:29:12:40:30:5D:59:B3:59:A4:F1:1D:24:C5:08:AF:F1:8D:E9:
> 		5E:B9:33:08:9A:6F:65:B0:E9:51:90:1F:66:A9:00:DE:3C:20:DB:EB:
> 		C7:BD:79:CE
> parm:           force_id:Override the detected device ID (ushort)
> parm:           fan_debounce:Enable debouncing for fan RPM signal (ushort)
> ```
> 
> But, when i try to load that module, it doesn't get loaded
> 
> ```
> modprobe nct6775
> modprobe: ERROR: could not insert 'nct6775': No such device
> ```
> 
> BTW, this is my kernel version
> 
> ```
> 4.18.0-513.18.1.el8_9.x86_64
> ```

Support of SuperIO used in your board was added to kernel last summer, I suppose such change is not backported to 4.18 LTS kernels, 6.4+ kernel version is required for your board

KeithMyers, 2024-04-12:
You are running a much too old kernel to have the newest nct6775 module that has code to recognize your motherboard SIO chipset. Likely your chipset is the nct6798 or 6799 chipsets which is only in the latest kernels.

You can try to force load the module with:

`sudo modprobe nct6775 force_id=0xd420`

Then run sensors again and see if the chipset is found at nct6799-isa-290.  If it is then add this to your /etc/modules-load.d/modules.conf file:
`options nct6775 force_id=0xd420`

If that doesn't work, then add this to your kernel command line:
`nct6775.force_id=0xd420`

and run sensors again.